andreknieriem/photobooth #230: Set Background Countdown, Outline Text Contdown Font

goema, 2020-02-29:
Allow to set a background URL for the coutdown.
So i can use motion as background as live-preview while the coutdown is going on.

Also helpful for this feature is an outline for the contdown font.

Best regards

andi34, 2020-03-01:
https://github.com/andi34/photobooth/commit/fe7a6c351b7c33d0785c4b427947c7906f7b1718

https://github.com/andi34/photobooth/commit/1d5f67bed823ca19607013e1698569b0a989dc80

andi34, 2020-03-01:
[photobooth-dev-SNAPSHOT.20200301083744205.zip](https://github.com/andreknieriem/photobooth/files/4271531/photobooth-dev-SNAPSHOT.20200301083744205.zip)

Wild-Bird, 2020-03-02:
Bei
`'placeholder' => 'url(localhost:8081)',`
sollte man noch ein protokoll mitgeben.
`'placeholder' => 'url(http://localhost:8081)',`
Was hab ich Depp den Fehler gesucht.

andi34, 2020-03-02:
> Bei
> `'placeholder' => 'url(localhost:8081)',`
> sollte man noch ein protokoll mitgeben.
> `'placeholder' => 'url(http://localhost:8081)',`
> Was hab ich Depp den Fehler gesucht.

Danke, werde ich ergänzen :)

andi34, 2020-03-02:
Wie sieht's aus mit
`'url(127.0.0.1:8081)',
Oder wird da auch das http benötigt?

Wild-Bird, 2020-03-03:
Mit 127.0.0.1 ist es genauso.