andreknieriem/photobooth #233: verschiedene PTP Errors

jnsn-b, 2020-03-19:
Hallo,

Ich habe das folgende Problem: Fotobox funktioniert ein paar Mal mit meiner Nikon D3200, dann auf einmal gibt es verschiedene PTP Error beim Auslösen per Konsole (LogFile unten).
Ich habe jetzt auch ein paar Mal schon gelesen, dass ich den GVFS Dienste killen oder deaktivieren soll, aber diese sind weder installiert noch am Laufen.

Gibt es Ideen?

Gruß 
Jens

Log:

pi@raspberrypi:~ $ sudo gphoto2 --debug --capture-image-and-download
0.000063 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000148 main                        (2): gphoto2 2.5.20
0.000178 main                        (2): gphoto2 has been compiled with the following options:
0.000210 main                        (2):  + gcc (C compiler used)
0.000268 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000295 main                        (2):  + exif (for displaying EXIF information)
0.000324 main                        (2):  + cdk (for accessing configuration options)
0.000350 main                        (2):  + aa (for displaying live previews)
0.000375 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000401 main                        (2):  + readline (for easy navigation in the shell)
0.000430 main                        (2): libgphoto2 2.5.22
0.000458 main                        (2): libgphoto2 has been compiled with the following options:
0.000485 main                        (2):  + all camlibs
0.000509 main                        (2):  + gcc (C compiler used)
0.000533 main                        (2):  + ltdl (for portable loading of camlibs)
0.000561 main                        (2):  + EXIF (for special handling of EXIF files)
0.000592 main                        (2): libgphoto2_port 0.12.0
0.000621 main                        (2): libgphoto2_port has been compiled with the following options:
0.000649 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000676 main                        (2):  + gcc (C compiler used)
0.000700 main                        (2):  + ltdl (for portable loading of iolibs)
0.000726 main                        (2):  + USB (libusb1, for USB cameras)
0.000751 main                        (2):  + serial (for serial cameras)
0.000775 main                        (2):  + no resmgr (serial port access and locking)
0.000802 main                        (2):  + no ttylock (serial port locking)
0.000827 main                        (2):  + no lockdev (serial port locking)
0.000853 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000878 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000903 main                        (2): invoked with following arguments:
0.000930 main                        (2):   --debug
0.000954 main                        (2):   --capture-image-and-download
0.001048 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001203 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.001386 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.001430 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0'...
0.001896 foreach_func                (2): Called for filename '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0/disk'.
0.002861 gp_port_library_list        (2): found fstab fsname proc
0.002916 gp_port_library_list        (2): found fstab fsname PARTUUID=738a4d67-01
0.002999 gp_port_library_list        (2): found fstab fsname PARTUUID=738a4d67-02
0.003347 gp_port_library_list        (2): found mtab fsname /dev/root
0.003413 gp_port_library_list        (2): found mtab fsname devtmpfs
0.003462 gp_port_library_list        (2): found mtab fsname sysfs
0.003504 gp_port_library_list        (2): found mtab fsname proc
0.003544 gp_port_library_list        (2): found mtab fsname tmpfs
0.003634 gp_port_library_list        (2): found mtab fsname devpts
0.003680 gp_port_library_list        (2): found mtab fsname tmpfs
0.003761 gp_port_library_list        (2): found mtab fsname tmpfs
0.003847 gp_port_library_list        (2): found mtab fsname tmpfs
0.003948 gp_port_library_list        (2): found mtab fsname cgroup2
0.003998 gp_port_library_list        (2): found mtab fsname cgroup
0.004038 gp_port_library_list        (2): found mtab fsname bpf
0.004127 gp_port_library_list        (2): found mtab fsname cgroup
0.004170 gp_port_library_list        (2): found mtab fsname cgroup
0.004211 gp_port_library_list        (2): found mtab fsname cgroup
0.004377 gp_port_library_list        (2): found mtab fsname cgroup
0.004421 gp_port_library_list        (2): found mtab fsname cgroup
0.004463 gp_port_library_list        (2): found mtab fsname cgroup
[... 495 lines cut ...]
0.962327 _cb_irq                     (2): Transfer 0x2296808 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.962368 _cb_irq                     (2): 0x2289550 with status 3
0.962397 _cb_irq                     (2): Transfer 0x2289550 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.962438 _cb_irq                     (2): 0x2289608 with status 3
0.962463 _cb_irq                     (2): Transfer 0x2289608 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.962499 _cb_irq                     (2): 0x2297f08 with status 3
0.962527 _cb_irq                     (2): Transfer 0x2297f08 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.962566 _cb_irq                     (2): 0x2297f90 with status 3
0.962595 _cb_irq                     (2): Transfer 0x2297f90 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.964023 gp_filesystem_reset         (2): resetting filesystem
0.964066 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.964090 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.964113 delete_all_folders          (2): Internally deleting all folders from '/'...
0.964138 lookup_folder               (2): Lookup folder '/'...
0.964165 lookup_folder               (2): Found! / is 0x2282cb8
0.964193 recurse_delete_folder       (2): Recurse delete folder 0x2282cb8//
0.964219 gp_port_free                (2): Freeing port...
0.964240 gp_port_close               (2): Closing port...
0.965033 gp_filesystem_reset         (2): resetting filesystem
0.965075 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.965107 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.965134 delete_all_folders          (2): Internally deleting all folders from '/'...
0.965161 lookup_folder               (2): Lookup folder '/'...
0.965189 lookup_folder               (2): Found! / is 0x2282cb8
0.965217 recurse_delete_folder       (2): Recurse delete folder 0x2282cb8//

andi34, 2020-03-19:
Warum per sudo?

Schau doch mal hier:
https://github.com/andreknieriem/photobooth/wiki/Installation-on-Debian

Insbesondere auch ab hier https://github.com/andreknieriem/photobooth/wiki/Installation-on-Debian#install-photobooth bezüglich gphoto.

closed 2020-03-19 by andi34

andi34, 2020-03-19:
Bei meiner D3000 und auch einer EOS 1000D muss ich grundsätzlich dem gphoto2 Volume Monitor die Ausführung verweigern, braucht aber danach einen reboot:
`sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor`

Wenn das nichts bringt, dann Gphoto deinstallieren:
`sudo apt-get purge gphoto2 libgphoto2-6`
Aktuelle Version installieren:
`wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && sudo bash gphoto2-updater.sh`
Aber auch hier ggf wieder an den gphoto2 Volume Monitor denken 

Da es ein qphoto Problem ist und kein Photobooth Problem im Zweifelsfall https://github.com/gphoto/gphoto2/issues durchstöbern oder dort ein Issue eröffnen wenn du die Instruktionen zur Installation alle befolgt hast und auch die aktuellste gphoto keine Besserung bringt.

andi34, 2020-03-19:
Kommt mir gerade in den Kopf: Live View ggf deaktivieren, ich meine öfter in der Vergangenheit gelesen zu haben das die live View bei manchen Geräten Probleme machen kann.

jnsn-b, 2020-03-20:
Hi @andi34,
vielen Dank für die Hinweise. Am Ende lag es aber wohl einfach an einem defekten USB-Kabel. Nach einem Tausch geht nun alles perfekt.
Gruß,
Jens

andi34, 2020-03-20:
Hehe, sehr gut :-) dann viel Spaß mit dem Projekt!