andreknieriem/photobooth #234: jpg filesize 0kb

f1ssl, 2020-03-22:
**Describe the bug**
I adjust all the settings to use the pi camera with this tool. if i start the command "raspistill -p -o /var/www/html/data/tmp/test.jpg -q 100 -t 1"with a specific filename it works fine and a jpg-file will be saved to /var/www/html/data/tmp

but if i want to make a picture via the website and use the %s instead of a filename, the filesize of the jpg is 0KB.

is there a problem with a specific user? andybody has an idea? thank you very much.

**To Reproduce**
Install everything via instruction and add the two commands into the config file for using the pi camera.

**Expected behavior**
a working photobooth

**Environment (please complete the following information):**
complete new setup, so newest raspbian buster lite

**Photobooth**
no
newest version

andi34, 2020-03-22:
As mentioned in the wiki:
```
$config['take_picture']['cmd'] = 'raspistill -n -o %s -q 100 -t 1 | echo Done';
$config['take_picture']['msg'] = 'Done';
```

andi34, 2020-03-22:
Also make sure the www-data User has to be part of the `video` group:

```
sudo gpasswd -a www-data video
reboot
```

f1ssl, 2020-03-22:
Thanks. Changed it in wrong config file, did it now in /var/www/html/lib/config.php an first try seems to work finde. Thank you.

andi34, 2020-03-22:
Great to here. Will be enough if you change it in the admin panel.

Closing the issue as the problem is solved.

closed 2020-03-22 by andi34