andreknieriem/photobooth #235: Live Preview while countdown fix

Piccar, 2020-04-02:
To get a Live view while the countdown is running you'll need a camera that can be used as a webcam and will not be controlled by Gphoto2. Simply get motioneye an mount the secound camera. Then open a stream on a Port you like. Then go into your Web folder and eddit the /config/my.config.inc.php.
You'll need to update the color of the countdown-background. 

should look something like this....
  array (
    'primary' => '#7fbe87',
    'secondary' => '#090300',
    'background_countdown' => 'url(http://yourlocaladress:8081)',

),

If you then also go into the /admin panel and set the background image also to url(http://yourlocaladress:8081) the illusion is complete.

NOTE: You can still eddit the live view in the motion eye control panel.

andi34, 2020-04-02:
If you like to use the admin panel for it you can add my implementation which can be controlled via admin panel.

https://github.com/andi34/photobooth/issues/44

andi34, 2020-04-02:
Requested here already https://github.com/andreknieriem/photobooth/issues/230

closed 2020-04-02 by andi34

Piccar, 2020-04-02:
whoops I'm sorry. Somehow I missed that!

andi34, 2020-04-02:
> whoops I'm sorry. Somehow I missed that!

Don't worry, https://github.com/andreknieriem/photobooth/issues/230 is written in german