andreknieriem/photobooth #236: How to set-up without USB connection to camera

fisk3110, 2020-04-08:
**Describe the bug**
Not directly a bug: I tried to cuse wifi remote control for my Sony Apla 7 II.
But I was not able to connect.

Is it in general possible to get your booth working with wifi cameras?

andi34, 2020-04-08:
You can use any program which can be used via command line. Not sure if gPhoto2 supports wifi connections.

Or is it about triggering the photo?
Therefore check https://github.com/andreknieriem/photobooth/blob/19780ee527f5d8d17666f3f18115665e797a2830/config/config.inc.php#L50-L53

Cheap Bluetooth/IR Presenter should work fine.

I am using this one:
[image]

closed 2020-04-08 by andi34

sualko, 2020-04-08:
According to a [german article](https://www.multimedia4linux.de/index.php/canon-dslr-wiki/wlan-wi-fi-mit-gphoto2-nutzen) this should be possible with `gphoto2 --port ptpip:CAMERA_IP`.