andreknieriem/photobooth #238: Use Raspberry Pi Cam in Preview and Capture

black-snake, 2020-05-08:
Hi,

first of all, thanks for your great project and please excuse me if the issue has already been raised, but somehow it hasn't been clearly answered.

I have the following problem with the Raspberry Pi Cam v2:

### case A
* use the camera with `device cam` for live preview
* use the camera with `raspistill -n -o %s -q 100 -t 1 | echo Done` for capture

**problems**
* sometimes capture fail unexpectedly (some here have reported the same problem)
* live preview only shows a zoomed part of the actual image captured

### case B
* use the camera with `device cam` for live preview
* use the camera with `device cam` for capture

**problems**
* all images have a low quality and the resolution of 1920 x 1080 pixels (no matter what I set in the admin interface; RPi Cam v2 can capture at 3280 × 2464 pixels)

Do you have any recommendations/ solutions to fix this?

Thanks in advance.

andi34, 2020-05-08:
https://github.com/andreknieriem/photobooth/wiki/FAQ#preview-by-device-cam

https://github.com/andreknieriem/photobooth/wiki/FAQ#can-i-use-a-live-stream-as-background

Low resolution of picture from preview is because it's capturing the picture directly from the preview canvas.

black-snake, 2020-05-08:
Ok, thanks for your answer.

Concerning case A, I modified my capture command to `sleep 0.25 && raspistill -n -o %s -q 100 -t 1 | echo Done`. So far, I haven't seen any capture failures but I will further test on that.

> But you need to make sure the resolution can be displayed on your display, else you'll see a zoomed preview

Preview was, indeed, due to a lower resolution of my display (it unexpectedly used a lower resolution than I thought). Now that I went down to the display resolution with the preview size, it works.

Thanks so far.

andi34, 2020-05-08:
Yeah, there could be an issue if it's switching to fast from the stream which is used to raspistill which means the camera is in use and raspistill fails because of that.

Thanks for the suggestion!

closed 2020-05-08 by andi34

black-snake, 2020-07-19:
Unfortunately, I have to reopen this issue because I am still experiencing the error described in case A sporadically.

Sometimes a bunch of images are taken without any problem and them several pictures in row fail (probably because `chromium` did not properly release camera resources). Also, I can even add 5s wait time between preview and `raspistill` and still have the error. Any idea how to solve this?

Is there any kind of command to force releasing all camera resources **before** calling `raspistill`?

andi34, 2020-07-19:
> 
> **I think this could be an easy fix to only react on key strokes (button presses) when not in capture mode.**

Here it is ** https://github.com/andi34/photobooth/commit/8d8424a02a9fd9a15484c9ea91290d208f423d82 **