andreknieriem/photobooth #240: [Wiki] Mistakte in Autostart File for Raspbian LXDE

black-snake, 2020-05-10:
Since I can't create a pull request for a wiki on GitHub (correct me if I'm wrong), a create a new issue here.

On the wiki page (https://github.com/andreknieriem/photobooth/wiki/FAQ#how-do-i-enable-kiosk-mode-to-automatically-start-photobooth-in-full-screen) it says:
> and add the following lines:

    @xset s off
    @xset -dpms
    @xset s noblank
    @chromium-browser --incognito --kiosk http://localhost/

However, for Chromium, it should be (no `-browser` after `chromium`):

    @chromium --incognito --kiosk http://localhost/

andi34, 2020-05-10:
Did this changed recently on chromium? Source?
Checking on Google I always find it the way it's mentioned in the wiki and not different.

black-snake, 2020-05-10:
Just tried it out and `chromium-browser` was not recognized as a command but `chromium` was.

For installation, I just typed `sudo apt install chromium`.

andi34, 2020-05-10:
Recommand way is `sudo apt install chromium-browser`.

`chromium-browser` is installed on Raspbian by default so we should keep it the way it is inside wiki.

closed 2020-05-10 by andi34

black-snake, 2020-05-10:
Oh, I'm sorry. I didn't realize that there are two different Chromium packages: `chromium` and `chromium-browser`.

Based on a short research, it seems that they differ in Google integration: `chromium-browser` seems to be Google-free whereas `chromium` is not. However, it was hard to find any comparison between both, tough.

tobiashaas, 2020-05-10:
@black-snake 

I don't think you understood the Kiosk Mode function.  There are no different Chromium packages.  With the command chromium browser, Chromium is addressed to start the browser in kiosk mode.  Nothing will be installed here, you just have to save the command as described in the wiki ;-) Read the wiki again carefully and you will find the error!

black-snake, 2020-05-10:
I know what the kiosk mode is (it is started with the flag `--kiosk`). My question was about the name of the executable and that my differs from the one in the wiki. But it seems as if there two (different) packages in the repos of the respective distorts (mine is Raspbian). I installed `chromium`, but there also seems to be a package called `chromium-browser` as well.