andreknieriem/photobooth #245: Countdown LED Buzzer

weili87, 2020-07-08:
Hallo zusammen,

erstmals vielen Dank für die Tolle Arbeit, ich finde das PhotoBooth echt klasse.

aber zu mein Problem, ich bin Neuling und hab noch nicht so viel Ahnung davon, bin gerade dabei es zu verstehen :-)
Ich versuche schon ein paar tage die Codezeilen einzufügen wo sie hin gehören. Ich finde aber nicht die richtige Datei und den richtigen Platz dazu. Kann mir von eich wer Helfen wo ich die Codezeilen einfügen muss?

> Countdown:
> 
> ```
>        if (start == 5) {
>                console.log('five');
>                var request = new XMLHttpRequest();
>                request.open(### "GET","http://10.0.0.2/BACK");
>                request.send();
>        }
>        if (start == 2) {
>                console.log('two');
>                var request = new XMLHttpRequest();
>                request.open(### "GET","http://10.0.0.2/TWO");
>                request.send();
>        } 
> ```
> 
> Regenbogen sobald das fertige Foto angezeigt wird
> 
> ```
>        if (data.error) {
>                    public.errorPic(data);
>                } else {
>                        if (photoStyle === 'collage') {
>                                console.log('COLLAGE LED');
>                                var request = new XMLHttpRequest();
>                                request.open(### "GET","http://10.0.0.2/COLLAGE");
>                                request.send();
>                        }else{
>                                console.log('PIC LED');
>                                var request = new XMLHttpRequest();
>                                request.open(### "GET","http://10.0.0.2/PIC");
>                                request.send();
>                        };
>                        public.renderPic(data.file);
>                }

Liebe Grüße
Markus

andi34, 2020-07-08:
Das ist eine Implementierung von @flighter18 
Einen Teil seines Quell-Codes hat er ja hier gepostet:
https://github.com/andreknieriem/photobooth/issues/198#issuecomment-561861099

Bezüglich der Umsetzung wird er dir vermutlich  am besten helfen können.

Ggf. Sieht die vore.js

andi34, 2020-07-08:
Sorry, falscher Knopf

flighter18, 2020-07-09:
@weili87 die Codezeilen kommen ins File resources/js/core.js

Ich habe dir mein core.js angehänt. So kannst du schauen, wo hin die Zeilen genau kommen :)
[core.js.txt](https://github.com/andreknieriem/photobooth/files/4895399/core.js.txt)

weili87, 2020-07-09:
Hallo @flighter18,
vielen dank für deine schnelle Antwort, ich werde es gleich mal versuchen.

LG Markus

weili87, 2020-07-11:
Hallo @flighter18 
Nochmal vielen Dank, es hat super funktioniert. 
Lg Markus

closed 2020-07-11 by andi34

flighter18, 2020-07-13:
@weili87  Schön dass es geklappt hat! :)

flacoonb, 2020-07-15:
@flighter18 
Hast du allenfalls eine kurze Anleitung, um deinen Buzzer nachzubauen oder würdest du so eine erstellen? Finde den genial, aber so ohne Anleitung... photobooth-statistic funktioniert dank deiner Hilfe auch einwandfrei!
Grüsse Philipp

flighter18, 2020-07-16:
@flacoonb 
Bin vor kurzem Vater geworden und habe somit keine Zeit um eine Anleitung zu erstellen 😅 😄 
Aber wenn du konkrete Fragen hast, werde ich dir die Fragen gerne beantworten. 

PS: Der Buzzer Code und die Liste mit der verwendeten Hardware ist schon im Git.

andi34, 2020-07-16:
Mal einfachheitshalber:
> 
> This are the parts i used for the buzzer:
> - https://www.play-zone.ch/de/adafruit-feather-32u4-bluefruit-le.html
> - https://www.play-zone.ch/de/polymer-lithium-ion-lipo-akku-1000mah.html
> - https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
> - https://www.digitec.ch/de/s1/product/delock-usb-20-a-micro-015m-20-usb-kabel-6089618
> - https://shop.griederbauteile.ch/product_info.php?cPath=1200_1223&products_id=20111
>

flighter18, 2020-07-16:
@andi34 
danke dir 😄

flacoonb, 2020-07-17:
@flighter18 
Kein Thema, das verstehe ich natürlich. Dann nachträglich herzliche Gratulation zur Vaterschaft!
Werde den Buzzer mal angehen, dann kann ich dann eine Anleitung dazu schreiben.

@andi34 @flighter18 
mit den Angaben kann ich doch schon was anfangen. Dank. 
Ich nehme an, der Buzzer kommunizert schliesslich über Bluetooth mit dem Pi. Muss der Pi, nebst Bluetooth, noch speziell konfiguriert werden?

flighter18, 2020-07-17:
@flacoonb 
Danke für deine Glückwünsche.

Ja genau, der Buzzer kommuniziert per Bluethooth mit dem PI. 
Auf dem Pi muss nichts gemacht werden

andi34, 2020-07-17:
Keycode für die Booth per Adminpanel definieren für Collage und Foto?
Von mir auch noch alles Gute und die besten Wünsche!

andi34, 2020-07-17:
Würde das passen @flighter18 ? https://www.amazon.de/hz/wishlist/ls/3CFZJ55F4NS05?ref_=wl_share

flighter18, 2020-07-17:
> Keycode für die Booth per Adminpanel definieren für Collage und Foto?
> Von mir auch noch alles Gute und die besten Wünsche!

Stimmt das musste ich auch noch anpassen.  Danke für den Hinweis und die Glückwünsche 😄

flighter18, 2020-07-17:
> Würde das passen @flighter18 ? https://www.amazon.de/hz/wishlist/ls/3CFZJ55F4NS05?ref_=wl_share

Ja sollte eigentlich auch passen. 
Jedoch finde ich, dass das Gehäuse relativ schmal ist und somit je nach dem zu wenig Platz hat für Akku und Board.

flacoonb, 2020-07-17:
Top, danke für die Infos, habe mir mal die Hardware bestellt. Werde mich bestimmt wieder melden, sobald ich zum "Basteln" komme.

flacoonb, 2020-07-28:
> @andi34 
> 
> danke dir 😄 
> 
> 
> 
> Der Link zum Akku ist nicht mehr aktuell. Es scheint so, als würde es diesen Akku nicht mehr geben. 
> 
> 
> 
> Ich denke dieser würde aber gehen:
> 
> https://www.bastelgarage.ch/lipo-akku-1000mah-jst-1-25-lithium-ion-polymer-fur-lora-ttgo?search=akku%201000mah
> 
> 
> 
> Wichtig ist einfach, dass er schon einen Stecker hat.

Der Stecker passt nicht, zu klein...
Auch der @andi34 hat einen Micro JST 1.25-Stecker und dürfte nicht passen.

https://www.amazon.de/dp/B08425TWKJ/?coliid=I21FX3BVS32OYJ&colid=35KKRDBJ9Z59S&psc=1

Der sollte passen.