andreknieriem/photobooth #248: Dropbox sync & hotspot access

florianmueller, 2020-07-17:
**Is your feature request related to a problem? Please describe.**
No. I implemented these features on the side in my photobooth project and thought I share it, as I found these to be very handy and I think they can be very easily integrated into the project.

**Describe the solution you'd like**
python Dropbox script that runs every 1 minute and syncs all images in data/images to a folder on dropbox. This way you can share the dropbox link before/while your party and everyone can access it. The python script would only need to be hooked to the admin interface settings, so you can enter your API key, desired target folder and enable/disable.
This is the script source and very easy to install: https://github.com/aicpp/cloudsync

Hotspot implementation was a bit more tricky, but having the option in the admin interface wo activate the hotspot and set wifi name and password would be very cool. The hotspot script needs a second wifi adapter, [like this RT5370 usb dongle](https://www.amazon.de/gp/product/B073J3HXZH/), but then it creates a wifi with a own subnet that gives you the option to connect to your photobox wifi and access the admin interface and dial into a new wifi when you are on the road with the box. No need to use a big router box or similar.
This is the script I used together with the dongle from amazon, works great!
https://github.com/rudiratlos/hotspot/blob/master/hotspot

**Describe alternatives you've considered**
-

**Additional context**
More details about how I installed it can be found at my projects documentation site:
https://florianmuller.com/raspberry-pi-photobooth-in-a-classic-vintage-plate-camera

andi34, 2020-07-18:
Hey Florian!
Good Idea!
I've seen you're using the SunFounder touch display, I've had to return mine as it had some scratches and burned in display (seems like someone used it before). Also I didn't liked it much. Now using a 5 inch display from Jun-Saxifragelec which works really good. (Both 5 and 7 inch are to small imo, should be 10 inch at least on a standalone Photobooth). You're happy with your display? Maybe I got the Monday version 🙈

About the hotspot:
We could add https://github.com/idev1/rpihotspot as dependencie ( https://github.com/jarettrude/photobooth/commit/4e0d46195e5a1097c5f14ecd2379bc1d7ba12db1#diff-7c3613dba5171cb6027c67835dd3b9d4 ) which would allow to setup a hotspot with one single line via Terminal:
```
sudo ./setup-network.sh --install-upgrade --ap-ssid="photobooth" --ap-password="password" --ap-password-encrypt
--ap-country-code="CA" --ap-ip-address="10.10.10.10" --wifi-interface="wlan0"
```
(Replace IP-Adresse, ssid and password to your own wishes).

Best regards
Andi

florianmueller, 2020-07-18:
Hehe, yes the SunFounder Touchscreen works great. I also find it very robust after quite some stress I put onto it. But I guess in general the rpi touchscreens are like most accessories for the pi, depending on try and error :) I had the UPERCT 7 inch screen (https://www.amazon.de/dp/B07L8CM5M8/) before, which failed twice in a short period of time. So despite its great reviews for me that screen was crap :) But yes I am really happy with the Sunfounder one. And agreed, display size should minimum be 10 inch for good usage, especially when using stuff like live view. But in my special case 7 inch was the perfect width to match the old camera case :) Still I had to add some nasty custom css to make it usable on this small screen, as the default design & buttons are way to small on the 7 inch and its resolution.

Maybe another idea for the admin panel to have a custom css input box that just adds the custom css to the app?😊

Cool idea with the hotspot setup. Maybe thats something to be integrated in the install script, where it asks you "Install hotspot yes no?" And if you don't input anything defaults to no.

jacques42, 2020-07-19:
On my RPi / iPad2 based photobooth I implemented a local WLAN hotspot for guests to be able to access a local stand-alone photobooth gallery, pointing to the exact data directory which my primary photobooth installation uses for taking pictures. That way guests can simply connect their mobile phones, access the library as it growth during the event and download pictures to their mobile phone right away. I used bind9 / isc-dhcpd instead of dnsmasq but either tools should do the job. And just to mention, the iPad2 I use as front-end touch screen uses that exact same WLAN hotspot of course, to connect to nginx.

The mentioned `rpihotspot` script seems to hard-code dnsmasq as DNS forwarder to Google DNS, which for me does not work, given that often I do not have Internet available access at the venue. In that use-case I need to have a DNS root setup locally, so that any client DNS queries resolves to the local RaspPi IP address, always. Also I can see that the script default uses hiding nat (iptables MASQ) which in my scenario for a stand-alone WLAN AP does not hurt but also is not required. Last, the script seems to enable ip_forwarding by default, which also for security considerations I don't think it is required. I very much like that script for the Raspbian and Raspberry Pi version compatibiliy, which is a great help of course. Long story short I believe this script is more geared to a scenario where the RaspPi has a local Ethernet link w/Internet connection in place, to setup the WLAN AP for to give clients full internet access by using the RPi as NAT enabled WLAN gateway router. This is a different use case than what I have in mind for my photobooth instance and use-case. 

Now, let me know if this is of interest and I am happy to invest time to build an encapsulated extension to the photobooth RPi setup script, to set up a WLAN AP on the local Pi, for use with a Photobooth RPi instance. And we can shape it as we go along.

andi34, 2020-07-23:
@jacques42 sure, if you have a better implementation. On my own fork I am using the rpihotspot too now which jarettrude committed on his fork. (https://github.com/andi34/rpihotspot maybe a starting point for you?))