andreknieriem/photobooth #253: USB webcam takes no image via webinterface but works fine via command line

Marvin-Hub, 2020-08-30:
Hi,

i would like to use a usb webcam and a raspberry pi to take photos instead of a DSLR.
However iam not able to make this happen.

**What i have tryed**
I used different command line tools such as fswebcam on the command line to make sure they work "by hand".

I changed the command at the localhost/admin to the according command such as "fswebcam -d /dev/video0 --save=%s". Additionally i tryed different other commands/options, everyone worked on the command line but none via the webpage. Alternativly i manually added the according exec command an api/takePhoto.php

I changed the permissions of data/images and data/tmp to 777 in order to make sure there is no permission problem

I made sure that exec works fine by adding exec("mkdir /var/www/html/data/tmp/foo") at https://github.com/andreknieriem/photobooth/blob/master/api/takePic.php#L38
This works and will make the dir when ever i try to make a photo via the webinterface.

I made sure that the $cmd is the same command as i tryed "by hand".

**Problem**
No file is created in data/images nor data/tmp. Therefore it fails as no file was created.
https://github.com/andreknieriem/photobooth/blob/master/api/takePic.php#L47

Iam very happy if one could help me out, i tryed plenty things.

andi34, 2020-08-30:
Does it work as www-data user?
```
sudo -u www-data -s
fswebcam -d /dev/video0 --save="/var/www/html/data/test.jpg"
```

If not: try adding www-data user for USB access:  
`sudo gpasswd -a www-data plugdev`

Now try again and test if it works.

Marvin-Hub, 2020-08-30:
> Does it work as www-data user?
> 
> ```
> sudo -u www-data -s
> fswebcam -d /dev/video0 --save="/var/www/html/data/test.jpg"
> ```
> 
> If not: try adding www-data user for USB access:
> `sudo gpasswd -a www-data plugdev`
> 
> Now try again and test if it works.

Thank you andi.

I tryed as you suggested:
```

www-data@raspberrypi:~/html/data/tmp$ fswebcam -d /dev/video0 --save="/var/www/html/data/test.jpg"
--- Opening /dev/video0...
Trying source module v4l2...
Error opening device: /dev/video0
open: Permission denied
Trying source module v4l1...
Error opening device: /dev/video0
open: Permission denied
Unable to find a source module that can read /dev/video0.
```
However adding the user to plugdev still shows the same error (even when changing the path to data/tmp/test.jpg)
But : `sudo gpasswd -a www-data video`
Will allow www-data to execute  `fswebcam -d /dev/video0 --save=/var/www/html/data/tmp/test.jpg
`
However using the webinterface still fails to create the file

andi34, 2020-08-30:
Great to hear!
Tried after a reboot ? (Sometimes magic for no reason)
Could you post your `my.config.inc.php` ?

andi34, 2020-08-30:
And maybe again make sure permissions are set right for www-data user  
`sudo chown -R www-data:www-data /var/www/`

Marvin-Hub, 2020-08-31:
> Great to hear!
> Tried after a reboot ? (Sometimes magic for no reason)
> Could you post your `my.config.inc.php` ?

Yes, i tryed reboots as well.
```

cat my.config.inc.php 
<?php
$config = array (
  'show_fork' => false,
  'cntdwn_time' => '1',
  'background_image' => 'url(../img/bg.jpg)',
  'background_admin' => 'url(../img/bg.jpg)',
  'background_chroma' => 'url(../img/bg.jpg)',
  'take_picture' => 
  array (
    'cmd' => 'fswebcam -d /dev/video0 --save=%s',
  ),
);
```

However, chown was the solution !!!
Thank you so much !

andi34, 2020-08-31:
No problem and good to hear it's working now👍
closing this issue as solved

closed 2020-08-31 by andi34