andreknieriem/photobooth #263: Blank Screen in Browser

Jhipsey, 2021-01-27:
After installing the software from github everything on the pi seems to work fine, but I got a white screen in every browser I use. Header tab says "Photobooth" and a symbol of a camera next to it though. 

How can I fix this.?
[image]

**Environment (please complete the following information):**
 - raspian lite (downloaded on January 27th, 2021)
 - each browser I tried (Chrome, IE, Firefox)

andi34, 2021-01-27:
Seems like you're missing the css files which get generated while running `yarn build`

Jhipsey, 2021-01-28:
Worked just fine today. Couldn´t find the problem, but it kind of solved itself

closed 2021-01-28 by Jhipsey