andreknieriem/photobooth #264: Couldn´t read JPEGS.

Jhipsey, 2021-01-28:
After the countdown the following mistake comes out. For me seems like the software can´t deal with the photo.

Is there a command to solve this problem?

[image]

**Environment (please complete the following information):**
 - Raspberry Pi OS Lite Kernel version 5.4
 - latest firefox browser
 - Rasperry cam V2

latest photobooth version: 2.1.0

andi34, 2021-02-02:
Setup like described here? https://github.com/andreknieriem/photobooth/wiki/FAQ#is-pi-camera-supported

andi34, 2022-10-04:
Please try on Photobooth v4. I am sure we'll find a solution if you still have trouble.
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34