andreknieriem/photobooth #268: http auf https umstellen

weili87, 2021-02-21:
Hallo,

ich hab ein Problem photobooth auf https umzustellen.
meine Frage dazu, kann mir da wer dabei helfen oder gibt es schon eine Anleitung dazu.
Wer echt nett wenn mir da einer Helfen könnte.

Mit freundlichen Grüßen
Markus

andi34, 2021-02-21:
Schau Mal hier:
https://lehrerfortbildung-bw.de/st_digital/medienwerkstatt/internet/cloud/zugriff/ssl.html

weili87, 2021-02-21:
vielen dank für die schnelle Antwort 

muss ich zwingen auf https umstellen wenn ich die Live Vorschau über gphoto2 haben will.

ich hör zwar wenn ich auf der Startseite bin das die Kamera auslöst, aber wenn ich ein bild mache kommt immer ein Fehler und Photo Booth startet neu. Wir könnte ich den Fehler beheben?

Ich nutze Debian mit einer Canon Kamera.

Lg Markus

andi34, 2021-02-21:
Die Funktion ist hier garnicht implementiert. Eine Implementierung dafür gibt es aber in meinem fork. Zur Referenz: https://github.com/andi34/photobooth/pull/131

Instruktionen findest du dafür in meinem Wiki.
Https wird nur benötigt, wenn du über die IP-Adresse aufrufst, über localhost bzw 127.0.0.1 nicht da diese automatisch als sichere Verbindung zählen.

closed 2022-10-04 by andi34