andreknieriem/photobooth #269: Buzzer an GPIO fuktioniert nicht......noch ein Issues, sorry

Fuenneff, 2021-03-23:
Hallo
Bei mir ist das gleiche Problem wie bei #128 von F-Major. Leider stand bei Ihm keine Lösung....
Ich hab einen Raspi 4 und die **2021-01-11-raspios-buster-armhf-full** Software installiert.
Nach einer Woche (kurz vorm durchdrehen) hab ich die Suche aufgegeben. 

Hab Andre's Installation genommen und die Buzzer installation von https://github.com/andreknieriem/photobooth/wiki/FAQ#can-i-use-hardware-button-to-take-a-picture-on-my-raspberry-pi
Hiermit **lsmod | grep uinput** bekomme ich auch was raus **uinput 20480 0** und wenn ich die beiden Kabel zusammen halte kommt **uinput 20480 1**
Nach eingabe von **python /var/www/html/button.py** kommt:
Traceback (most recent call last):
  File "/var/www/html/button.py", line 17, in <module>
    with uinput.Device([uinput.KEY_ENTER]) as device:
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 178, in __init__
    self.__uinput_fd = fd or fdopen()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 84, in fdopen
    return _libsuinput.suinput_open()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 70, in _open_error_handler
    raise OSError(code, msg)
OSError: [Errno 13] Failed to open the uinput device: Permission denied

Bin echt ratlos. hab auch im Adminpanel mal die 18 gewählt....irgendwo ist ein Fehler nur finde ich den nicht.
Die Photobooth ist eine Super Idee und gefällt mir richtig gut. Vor allem muss man die Fotos nicht  an alle verteilen, weil Sie die selbst gleich runterladen können!

Freu mich von Euch zu hören/lesen ;-)

andi34, 2021-03-23:
Hey @Fuenneff,

Versuch doch gerne Mal die Implementierung aus meinem Fork. Hier lässt sich alles via Adminpanel regeln, die button.py ist in der aktuellen development Version sogar hinfällig geworden.

https://github.com/andi34/photobooth

Viele Grüße

Fuenneff, 2021-03-24:
HI @andi34,
vielen Dank für die schnelle Antwort. Ich fang gleich mal an den Raspi neu aufzusetzen. Ich melde mich wieder.

Viele Grüße

andi34, 2021-03-24:
Kein Thema :) bei Problemen oder Fragen kannst du dich auch gerne in der Telegram Gruppe melden. Der Installer übernimmt alle wichtigen Einstellungen im system. Wichtig ist, dass du die development Version installierst.
Für den remotebuzzer Server ist es dann wichtig, dass du im Adminpanel die IP-Adresse des Pi hinterlegst.
Unter http://localhost/manual/faq.php findest du auch ein auf die installierte Version angepasstes FAQ.

Fuenneff, 2021-03-26:
Hallo @andi34,
ich hab die neue Photobooth vorgestern installiert und mal neben bei super geile arbeit!
Nun ist das so, mit dem Buzzer: 
Im Entwicklungs-Modus funktioniert es.
remotebuzzer_server.log:

Remote Buzzer Server [3295]: Server starting on http://192.168.0.199:
Remote Buzzer Server [3295]: socket.io server started
Remote Buzzer Server [3295]: Operating in BUTTON mode
Remote Buzzer Server [3295]: Connecting Picture Button to Raspberry GPIO 21
Remote Buzzer Server [3295]: Initialization completed
Remote Buzzer Server [3295]: PID file created [ data/tmp/remotebuzzer_server.pid ]
Remote Buzzer Server [3295]: GPIO 21 - Picture button pressed
Remote Buzzer Server [3295]: GPIO 21 - Picture button released - normal - 51  [ms]
Remote Buzzer Server [3295]: Photobooth trigger PICTURE : [ photobooth-socket ] => [ All Clients ]: command [ picture ]

Wenn ich aus den Entwicklungs-Modus raus gehe, steht folgendes unten dran
Service Control [ config ]: Photobooth config has changed, killed processes by name remotebuzzer_server.js -> PID 3295
Service Control [ config ]: Photobooth config has changed, killed processes by name remotebuzzer_server.js -> PID 3325

In der /boot/config.txt ist nur das activiert:
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
**dtoverlay=vc4-fkms-v3d**
**max_framebuffers=2**

[all]
#dtoverlay=vc4-fkms-v3d
# Photobooth
**gpio=16,20,21,26=pu**
# Photobooth End

Hast Du eine Idee für mich?

andi34, 2021-03-26:
Mal den Pi neu gestartet?

Fuenneff, 2021-03-26:
Hab ich ....nichts passiert.

andi34, 2021-03-26:
Frisch über den Installer installiert oder per Hand?
```
wget https://raw.githubusercontent.com/andi34/photobooth/dev/install-raspbian.sh
sudo bash install-raspbian.sh
```
(`Install Last development Version` bei Rückfrage)
Ggf schau doch Mal in unserer Telegram Gruppe vorbei.
https://t.me/PhotoboothGroup

Fuenneff, 2021-03-26:
Raspi OS neu installiert und über den Installer installiert.
ok, dann ist das mit den Kill ein wenig komisch....

andi34, 2021-03-26:
Ggf. Mal den remotebuzzer Server deaktivieren, config speichern, erneut aktivieren und wieder speichern.
Komme gerade nicht an meinen Pi ran.

Fuenneff, 2021-03-26:
Ne, leider auch nicht. Mal ganz doof nachgefragt.... soll bei Remote Buzzer Server ein PORT eingetragen werden?

andi34, 2021-03-26:
Ja, default ist Port `14711`.

Wie schaut deine `config/my.config.inc.php` aus?

Fuenneff, 2021-03-26:
14711 steht in dem Feld drin. Hier die Config:

<?php
$config = array (
  'ui' =>
  array (
    'language' => 'de',
    'show_fork' => false,
    'style' => 'classic',
  ),
  'adminpanel' =>
  array (
    'view' => 'expert',
  ),
  'start_screen' =>
  array (
    'title' => 'Mottoparty',
  ),
  'webserver' =>
  array (
    'ip' => '192.168.0.199',
  ),
  'delete' =>
  array (
    'no_request' => true,
  ),
  'event' =>
  array (
    'textLeft' => 'Blau kommen',
    'symbol' => 'fa-users',
    'textRight' => 'Blau gehen',
  ),
  'filters' =>
  array (
    'enabled' => false,
  ),
  'collage' =>
  array (
    'enabled' => false,
    'continuous' => false,
    'limit' => 4,
  ),
  'gallery' =>
  array (
    'enabled' => false,
    'use_slideshow' => false,
    'allow_delete' => false,
  ),
  'pswp' =>
  array (
    'history' => false,
  ),
  'preview' =>
  array (
    'mode' => 'device_cam',
    'url' => '192.168.0.199:8081',
  ),
 'mail' =>
  array (
    'subject' => 'Hier ist dein Bild',
    'text' => 'Hey, dein Bild ist angehangen.',
  ),
  'remotebuzzer' =>
  array (
    'enabled' => true,
    'collagetime' => '4',
    'port' => '',
  ),
  'background' =>
  array (
    'defaults' => 'url(/resources/img/bg_bluegray.jpg)',
    'admin' => 'url(/resources/img/bg_bluegray.jpg)',
    'chroma' => 'url(/resources/img/bg_bluegray.jpg)',
  ),
);

Fuenneff, 2021-03-26:
Ich hab es gerade auch erkannt.....da steht kein Port drin in der config und in dem Admin Bereich ist das dann wohl eine "Geisterzahl"
Hab jetzt den Port 14711 eingetragen und schon klappt es.
Ich freu mich riesig!!!!!!!!!!!!!!!!!!

Danke für Deine Hilfe.....Kaffee kommt ;-)

closed 2021-03-26 by Fuenneff

andi34, 2021-03-26:
Super! Freut mich zu hören 😀 dann schonmal viel Spaß mit der Photobooth 👍