andreknieriem/photobooth #271: Passport Photo Feature

S1M0NDOTCOM, 2021-04-10:
I think that's a good feature. in addition, the project would have a larger target group

fotokrauskopf, 2021-05-05:
Ihr wollt Euch nicht mit den Reklamationen der abgelehnten Bilder aus den Ämtern herum schlagen... Viele Grüße Marcel

andi34, 2021-05-05:
@fotokrauskopf a 2x4 layout should be easy to add and I thought the same you told. Calling it "passport" should only mean the single picture size on the "collage" and not for official passport pictures.

I am about to look at it once I've finished working on a Photobooth integrated updater on my fork.

andi34, 2022-10-04:
A custom collage Layout can be defined in Photobooth V4
https://photoboothproject.github.io/FAQ#how-can-setup-a-custom-collage-design

closed 2022-10-04 by andi34