andreknieriem/photobooth #276: Picture menu disappears after entering the gallery

mhellmeier, 2021-07-03:
**Describe the bug**
After a picture is taken and the gallery is viewed, the bottom menu bar disappears.

**To Reproduce**
Steps to reproduce the behavior:
1. Take a picture
2. Click on the Gallery button on the lower left
3. Scroll through the gallery and close it
4. See the error: The bottom menu disappears

**Expected behavior**
Menu not faded down.

**Environment (please complete the following information):**
 - OS: Raspbian
 - Browser chromium-browser
 - Version 2.10 (latest)

**Additional context**
When reviewing the code, the normal behavior is made through the following CSS:
```
.resultInner.show {
    bottom: 0;
}
```
When coming back from the gallery to the picture, the `.resultInner` is used with a `bottom: -64px;` that causes the problem.

andi34, 2021-07-04:
Doesn't it show up again on a click somewhere on the screen?

mhellmeier, 2021-07-04:
Yes, it shows up again after tapping the screen, but it should come back directly after the gallery is closed. This is a much more user-friendly solution.

commit 8f4d66bc (2021-07-05): Show picture menu after gallery closes (fixes #276)
closed 2021-07-05 by andreknieriem