andreknieriem/photobooth #278: Doesn't work with node version 16

mhellmeier, 2021-07-04:
The application can only be built with node version 10.x. When trying to build it with the latest node version (16.x), it fails because the node version doesn't match the `node-sass` version:
```
error:  /path/to/photobooth/node_modules/node-sass: Command failed
```

Have a look at [this SO post](https://stackoverflow.com/questions/60394291/error-node-modules-node-sass-command-failed) for more information.

andi34, 2021-07-06:
You're using yarn as suggested?

andi34, 2021-07-06:
For Node >=V12 https://github.com/andi34/photobooth/pull/264

mhellmeier, 2021-07-07:
> You're using yarn as suggested?

Yes of course. But the latest yarn version is shipped with node 16.x.

andi34, 2022-10-04:
The new Photobooth Installer handles needed node Version.
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34