andreknieriem/photobooth #283: Absolute path in config

Matz3187, 2021-07-06:
Hello Guys,

I have a problem with photobooth. I want so save pictures on USB drive. When I change the image path in config to an absolute path "/media/usbdrive" (mounted USB drive). photobooth will create a folder in wwwroot named media and inside folder usbdrive.

Is there any posibility to set image path to an absolute path.

Kind Regards
Matthias

andi34, 2021-07-06:
Not tested, but might work.
```
sudo mv  /var/www/html/data/images  /var/www/html/data/images-bak
ln -s /var/www/html/data/images /media/usbdrive/
sudo chown -R www-data:www-data /var/www/
sudo chown -R www-data:www-data /media/usbdrive/
```

Else the file sync feature inside my Photobooth v3 release might be interesting for you: https://github.com/andi34/photobooth/releases/tag/v3.0.0

Matz3187, 2021-07-09:
Thank you :)

This solution worked for me.

Kind Regards 
Matthias

closed 2021-07-09 by Matz3187