andreknieriem/photobooth #284: External Button on Raspberry with Tablet as Display

Matz3187, 2021-07-09:
Hello :)

I'm using an Android Tablet for showing the Webinterface which is hosted on a Raspberry Pi.
The Solution with the hardware button python script requires that the webinterface is shown directly on Raspberry.
I want to use an external Button which is connected on a GPIO pin on Raspberry and also want to use the Tablet for display.

My Idea:
- Script on Raspberry check continously for Button press (like actual script in root)
- If the button is pressed, the script creates a file in www-root like "buttonpressed" and change owner to www-data
- The webinterface has a timer (interval 1sec for example) which is checking if file "buttonpressed" is available.
- Then webinterface starts countdown function and delete file "buttonpressed".

Do you think this is possible?

Kind Regards
Matthias

andi34, 2021-07-09:
Remotebuzzer Implementation was added on my fork already: https://github.com/andi34/photobooth/releases

Matz3187, 2021-07-09:
Is it possible in your fork to start countdown on the Tablet when the remotebuzzer is connected to the raspberry and not to the tablet?

andi34, 2021-07-09:
Yes.
https://github.com/andi34/photobooth/wiki/FAQ#can-i-use-hardware-button-on-my-raspberry-pi-to-take-a-picture

TeyBay, 2021-09-14:
I Have problems with the hardware button. I pull GPIO21 to ground but no picture is taken. How to connect the Button?

andi34, 2021-09-14:
Depends on your Photobooth version and maybe on how you're accessing your Photobooth (display connected on a Pi? Tablet?).

On v2.1
You need to follow
https://github.com/andreknieriem/photobooth/wiki/FAQ#can-i-use-hardware-button-to-take-a-picture-on-my-raspberry-pi ; 
- GPIO24 used to send a key code "13" (enter) while pressing the button
- you need to add the key code "13" inside the Adminpanel to trigger a picture (for reference see https://github.com/andreknieriem/photobooth/blob/fc67c61017c815a902cddc427b00efcc61119ef0/config/config.inc.php#L50-L53 )
- you must access Photobooth on a display connected to your raspberry, does not work on remote displays (e.g. Tablet)

On v3.x:
Follow https://github.com/andi34/photobooth/blob/dev/faq/faq.md#can-i-use-hardware-button-on-my-raspberry-pi-to-take-a-picture ; thanks to the remotebuzzer you can access Photobooth also on a different device (e.g. Tablet) and the button takes affect.

TeyBay, 2021-09-14:
I have a display directly connected and I use version 3.x. But in your tutorial there is gpio21 named as picture gpio. And what about the gpio0 to ground? I am so lost :D

TeyBay, 2021-09-14:
I got this error: Remote Buzzer Server [1020]: buttonActiveCheck error state - requested GPIO  21 , for value  0 but buttonIsPressed: 21

TeyBay, 2021-09-14:
Got it, with an pull-up resistor to 3.3V it works fine. Sorry.