andreknieriem/photobooth #289: chmod: cannot access '/usr/lib/gvfs/gvfs-gphoto2-volume-monitor': No such file or directory

noisestream, 2021-08-24:
**Describe the bug**
when running the install script 'install-raspbian.sh' I get this message.
Also when trying to install directly from git.
When trying to take a picture I get the message 'gphoto returned with an error code'.

**To Reproduce**
Steps to reproduce the behavior:
1.  sudo bash install-raspbian.sh
2. yes
3. no or yes
4. Open the Browse, can see the site, take Pic --> noting work.

**Environment (please complete the following information):**
 - OS: "Raspbian GNU/Linux 10 (buster)"
 - last Version
 - Raspi PI 4 B

**Photobooth**
- Did it work before? no - new install
- Which version of Photobooth are you using? 2.1.0

Thx very much. I need it for this weekend. :-)

andi34, 2021-08-24:
You might want to join https://t.me/PhotoboothGroup ? I bet we'll find a solution

andi34, 2021-09-09:
Issue solved?

noisestream, 2021-09-10:
The message still appears when running the installation script
"chmod: cannot access '/usr/lib/gvfs/gvfs-gphoto2-volume-monitor': No such file or directory" 
But the photo station works.
I close the issue

closed 2021-09-10 by noisestream