andreknieriem/photobooth #292: Changing color for textonprint

danielovic295, 2021-09-18:
**Is your feature request related to a problem? Please describe.**
When using the Photobooth and setting the text on print on the lower areas of the picture, the black printing will not be seen very good on the picture, as the people wearing dark pants or skirts.

**Describe the solution you'd like**
An easy option to change the font colour via admin panel.

**Describe alternatives you've considered**
Setting a light frame, but this always scales down the size of the picture. 

**Additional context**
Nothing.

andi34, 2021-09-18:
https://github.com/andi34/photobooth/commit/dd285e4255b144b82f12cf285da04465c5c10edf

Gives you the possibility to choose between white, grey and black.

andi34, 2021-09-18:
To choose any color you like: https://github.com/andi34/photobooth/commit/530a5f223349b57a9ee24c7665973c708f2fc85c

andi34, 2022-10-04:
Requires Photobooth v3.3.0 or later
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34