andreknieriem/photobooth #295: Instant print?

henry76, 2021-10-03:
Hi, 

I am looking for a way to print the collage after it has been taken. I tried adding a single sleep/print command to the print command, but for a collage it will not work. 
The aim is to avoid people hitting the touchscreen later that night :)...

Any clues on how to do that? 

BR, Henry

andi34, 2021-10-03:
I've added this to my fork https://github.com/andi34/photobooth/commit/2ae06404f7bd05ac1ce5e2402b1381c00f5906d0

henry76, 2021-10-03:
> I've added this to my fork [andi34@2ae0640](https://github.com/andi34/photobooth/commit/2ae06404f7bd05ac1ce5e2402b1381c00f5906d0)

Hi, thank you. Added the lines to version 2.10. Works like a charm 👯  after changing 
api.printImage(filename); to public.printImage(filename);
in src/js/core.js

andi34, 2021-10-03:
Yeah, I've forgot to mention. might need adjustments to work on v2.x, I've changed some more things too.

closed 2021-10-03 by andi34