andreknieriem/photobooth #299: mobile Download Button in Gallery funktioniert nicht

HappySeppel, 2021-12-29:
Hi,
wenn ich die Gallery als Standalone über das Handy  aufrufe bekomme ich beim Download der Bilder nur die Thumbs heruntergeladen aber nicht die eigentlichen "orginalen" Bilder. Rufe ich die Website über ein Notebook auf funktioniert der Download, wie kann ich das Problem lösen?
[image]

Vielen Dank im Voraus

andi34, 2021-12-29:
Problem deines Browsers?
Je nach Photobooth--Version ist es eine Option im Adminpanel ( v2.10.0 und neuer).

HappySeppel, 2021-12-29:
Hi,
ich habe sowohl Firefox als auch Chrome ausprobiert, jedes mal das gleiche.
Current version: 2.1.0
Welche Option meinst du im Admin Panel?

andi34, 2021-12-29:
Dann ist deine Version zu alt. Wie gesagt ab v2.10.0 lässt sich einstellen ob thumbails beim Download verwendet werden sollen oder nicht.

[image]

Dann hakt es bei dir in der Linkerstellung irgendwo, vermutlich hier:
https://github.com/andreknieriem/photobooth/blob/13e779d367f8c7937ca7289120e078594b8a53ae/resources/js/photoinit.js#L123-L130

HappySeppel, 2021-12-29:
Danke für deine Unterstützung. Ich habe in der download.api eine Zeile eingefügt welche mit per regex die fehlerhafte "thumbs" Erweiterung entfernt.
Da ich den Code etwas angepasst habe für meine Gallery Only Funktion, kann ich nicht ohne weiteres auf die höhere Version springen. Aber ich werde es bei Gelegenheit probieren.

closed 2021-12-29 by HappySeppel

HappySeppel, 2021-12-29:
Hi ich muss nochmal kurz stören. Du hast geschrieben Version 2.10 ist verfügbar, ich sehe nur die Version 2.1.0 siehe Screenshot. 
[image]

andi34, 2021-12-29:
https://github.com/andi34/photobooth/releases