andreknieriem/photobooth #300: External Buttons - Raspberry - Print and QR Code

FMarceloS, 2022-02-13:
Hello :)

Congratulation on this awesome project! Just installed everything on my Raspberry and it works very well.

Would be great if there would be the possibility to use it without a touch display. I only have a display without touch. Is it possible to add an external button for the printer and the QR Code for the download?

Thank you and kind regards!
Marcelo

andi34, 2022-02-13:
You can use a Rotary Controller, implemented here:
https://github.com/andi34/photobooth

Documentation can be found at:
https://github.com/andi34/photobooth/blob/wiki/FAQ.md#remotebuzzer-hardware-button-feature-using-gpio-connected-hardware-raspberry-pi-only

FMarceloS, 2022-02-13:
Nice, thank you!

I already installed the https://github.com/andreknieriem/photobooth

Do I need to uninstall it or can I just install the https://github.com/andi34/photobooth ?

andi34, 2022-02-13:
Clean installation is recommended since a lot of changes have been made.