andreknieriem/photobooth #321: Sony Camera not working propper

AndreasSteibl, 2024-02-17:
i have a question where could i find more info for a not propper working camera?

From starting i get 10 message per seconds with "(ptp_usb_getresp [usb.c:516]) PTP_OC 0x1009 receiving resp failed: PTP Access Denied (0x200f)"

Taking a photo, the first goto works but every seconds photo don't (the connection to the server is lost and afeter some seconds i can make a photo again this works, the next wont)

There are so many log but i think this the wrong part
`2024-02-17 20:46:56,582 [    INFO] (gp_context_error) Sorry, your Sony camera does not seem to return a JPEG image in LiveView mode (port_log.py:20)
2024-02-17 20:46:56,582 [    INFO] (gp_camera_capture_preview [gphoto2-camera.c:1388]) 'camera->functions->capture_preview ( camera, file, context)' failed: -1 (port_log.py:20)
2024-02-17 20:46:56,582 [ WARNING] error capturing frame from DSLR: [-1] Unspecified error (gphoto2.py:212)`

andi34, 2024-02-17:
Your issue seems not related to this project, besides that it's discontinued here.

Install latest development version of gphoto2 and libgphoto:
```
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env
chmod +x gphoto2-updater.sh
sudo ./gphoto2-updater.sh -d
```
Maybe needed:
```
sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
reboot
```

Please to report issues with gphoto2:
https://github.com/gphoto/gphoto2
(Take a look at the issues and use the search function before reporting a new issue there)
Also worth taking a look athttps://github.com/gphoto/libgphoto2

closed 2024-02-17 by andi34

AndreasSteibl, 2024-02-17:
I allready tested it qith gphoto2 `gphoto2 --capture-image` this wirks fine for as many image i would like, but in photobooth it didn't work.
Therefor i posted this issue here how to find the problem

andi34, 2024-02-17:
Capture isn't capture and download. Photbooth expects one jpg file for a single capture.

AndreasSteibl, 2024-02-17:
Ah okay i thought because the command works fine it should also in the app 
Can you say which command i can test in the command line  which should be similar to the app?

andi34, 2024-02-18:
https://photoboothproject.github.io/

There's install instructions for latest Photobooth, also the FAQ contains possible solution and ways to find out what might be the problem for you https://photoboothproject.github.io/faq/#gphoto2-troubleshooting

But your log message posted in OP seems like you're using a different Photobooth application.
https://github.com/photobooth-app/photobooth-app/
With documentation at https://photobooth-app.org/ and discussions at https://github.com/photobooth-app/photobooth-app/discussions

AndreasSteibl, 2024-02-18:
Ohhh okay, my fault - now i have the correct app.

The process of image taking works now as expected.
One problem i have:
I didn't get the live image from the DSLR to work.
I would like to see the live image on the startpage, before i take a photo.
But at the moment, i have no external monitor on the RP5 - should the live preview from DSLR should work on my remote Notebook?

On the admin i set: Preview mode : from device cam (no remote)
Command to generate live Preview: python3 cameracontrol.py --bsmtime 1
Execute Start command for preview: On

andi34, 2024-02-18:
Like said, the project is discontinued here and also GitHub is the wrong place for personal support. Everything important is mentioned in the FAQ regarding preview and it's options and limitations. Please take your time to read it (mentioned: cameracontrol.py --> no remote preview,
Go2rtc setup.variant & preview from URL--> remote preview possible).

If you're still having trouble feel free to join the community on telegram.